frequenz-floss/frequenz-sdk-python #650: Distinguish assets and devices in glossary

cwasicki, 2023-09-08:
### What's needed?

The glossary is not distinguishing devices and assets in the component section. Follow up on https://github.com/frequenz-floss/frequenz-sdk-python/pull/647#discussion_r1318691630

llucax, 2023-09-08:
What do you suggest is having different section for each type of those types or to have a definition for **Asset**? A more concrete proposal would be helpful.

cwasicki, 2023-09-08:
Assigned it to myself and create a proposal PR after your one is merged.

llucax, 2024-04-09:
@cwasicki is this done?

cwasicki, 2024-05-22:
Started drafting it here: https://github.com/cwasicki/frequenz-sdk-python/pull/1. Will create a PR for this to upstream and ping again.

cwasicki, 2024-05-22:
https://github.com/frequenz-floss/frequenz-sdk-python/pull/952

commit 40874140 (2024-06-24): Update glossary (#952)

Adding consumer and distinction between assets and devices.

Fixes #650
PR #952 (2024-06-24): Update glossary
Adding consumer and distinction between assets and devices.

Fixes #650
closed 2024-06-24 by cwasicki