frequenz-floss/frequenz-sdk-python #662: `tests/microgrid/test_client.py` is flaky

llucax, 2023-09-12:
### What happened?

Lately we failures on `tests/microgrid/test_client.py` very often. In particular:

```
tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_inverter_data FAILED [ 35%]
tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_ev_charger_data FAILED [ 36%]
tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_charge FAILED [ 36%]
tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_discharge FAILED [ 36%]
tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_set_bounds FAILED [ 36%]
```

### What did you expect instead?

Test should always pass.

### Affected version(s)

v0.25ish

### Affected part(s)

Microgrid (API, component graph, etc.) (part:microgrid), Unit, integration and performance tests (part:tests)

### Extra information

`pytest` logs for the failure:

```
=========================== short test summary info ============================
FAILED tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_inverter_data - grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Failed to list components. Microgrid API: [::]:57899. Err: failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused"
	debug_error_string = "UNKNOWN:failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused {created_time:"2023-09-12T08:55:10.296421026+00:00", grpc_status:14}"
>
FAILED tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_ev_charger_data - grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Failed to list components. Microgrid API: [::]:57899. Err: failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused"
	debug_error_string = "UNKNOWN:failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused {created_time:"2023-09-12T08:55:10.390244461+00:00", grpc_status:14}"
>
FAILED tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_charge - grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Failed to set power. Microgrid API: [::]:57899. Err: failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused"
	debug_error_string = "UNKNOWN:failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused {created_time:"2023-09-12T08:55:10.422726408+00:00", grpc_status:14}"
>
FAILED tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_discharge - grpc.aio._call.AioRpcError: <AioRpcError of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Failed to set power. Microgrid API: [::]:57899. Err: failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused"
	debug_error_string = "UNKNOWN:failed to connect to all addresses; last error: UNKNOWN: ipv6:%5B::%5D:57899: Failed to connect to remote host: Connection refused {created_time:"2023-09-12T08:55:10.484575096+00:00", grpc_status:14}"
>
FAILED tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_set_bounds - assert 4 == 0
 +  where 4 = len([target_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n, component_id: 1\ntarget_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n, component_id: 2\ntarget_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n, component_id: 3\ntarget_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n])
 +  and   0 = len([])
 +    where [] = <bound method MockMicrogridServicer.get_bounds of <tests.microgrid.mock_api.MockMicrogridServicer object at 0x7fde85ebc150>>()
 +      where <bound method MockMicrogridServicer.get_bounds of <tests.microgrid.mock_api.MockMicrogridServicer object at 0x7fde85ebc150>> = <tests.microgrid.mock_api.MockMicrogridServicer object at 0x7fde85ebc150>.get_bounds
================= 5 failed, 360 passed, 10 warnings in 47.42s ==================
```

```
=================================== FAILURES ===================================
__________________ TestMicrogridGrpcClient.test_inverter_data __________________

self = <frequenz.sdk.microgrid.client._client.MicrogridGrpcClient object at 0x7fde865fc050>

    async def components(self) -> Iterable[Component]:
        """Fetch all the components present in the microgrid.
    
        Returns:
            Iterator whose elements are all the components in the microgrid.
    
[... 539 lines cut ...]
            target_metric = microgrid_pb.SetBoundsParam.TargetMetric
            expected_bounds = [
                microgrid_pb.SetBoundsParam(
                    component_id=comp_id,
                    target_metric=target_metric.TARGET_METRIC_POWER_ACTIVE,
                    bounds=metrics_pb.Bounds(lower=-10, upper=2),
                )
                for comp_id in range(num_calls)
            ]
            for cid in range(num_calls):
                await microgrid.set_bounds(cid, -10.0, 2.0)
                await asyncio.sleep(0.1)
    
        finally:
            assert await server.graceful_shutdown()
    
>       assert len(expected_bounds) == len(servicer.get_bounds())
E       assert 4 == 0
E        +  where 4 = len([target_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n, component_id: 1\ntarget_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n, component_id: 2\ntarget_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n, component_id: 3\ntarget_metric: TARGET_METRIC_POWER_ACTIVE\nbounds {\n  lower: -10\n  upper: 2\n}\n])
E        +  and   0 = len([])
E        +    where [] = <bound method MockMicrogridServicer.get_bounds of <tests.microgrid.mock_api.MockMicrogridServicer object at 0x7fde85ebc150>>()
E        +      where <bound method MockMicrogridServicer.get_bounds of <tests.microgrid.mock_api.MockMicrogridServicer object at 0x7fde85ebc150>> = <tests.microgrid.mock_api.MockMicrogridServicer object at 0x7fde85ebc150>.get_bounds

tests/microgrid/test_client.py:605: AssertionError
```

shsms, 2023-09-13:
I'm wondering if it is because we're running some tests parallel now.  When pytest_min and pytest_max were running one after the other it was all fine.

I guess the nicer solution is to just do a complete rewrite of the mock_api from mocking a service to mocking a client.

But a quick solution can be to just use different ports and check if they are free before starting the mock api.

llucax, 2023-09-14:
They should run in different machines (virtual or not, GitHub should handle the load of their physical machines), so I doubt it is that, but who knows :shrug:

llucax, 2023-09-14:
Ah, tests seem to be failing in for the cross-arch test, things are definitely much slower inside QEMU.

llucax, 2023-09-14:
No, they also fail for the normal `nox` tests:

[image]

https://github.com/frequenz-floss/frequenz-sdk-python/actions/runs/6186797412/job/16795233606

llucax, 2023-09-19:
I've been doing some tests locally and this is definitely not related to running `pytest_min` and `pytest_max` in parallel, I can reproduce this more or less easily locally (maybe 1 in 2 or 3 runs fail locally) when running with `nox` (running `pytest` directly). Nox is slower, so I guess this is just another timing issue with the server not starting up in time or something.

Even at least some of the failures seen in #654 are the same as this, so maybe it just happens that the GitHub runners for Mac are slow: https://github.com/frequenz-floss/frequenz-sdk-python/actions/runs/6146555906/job/16676164697?pr=654#step:8:1569

Looking a bit more into it, I see in the logs that when getting component data, there are a lot of [retries](https://github.com/frequenz-floss/frequenz-sdk-python/blob/2c8f61f6cab08d77a837fb0e730c4887ff678c4f/src/frequenz/sdk/microgrid/client/_client.py#L335-L355) [^1]. When listing components, [the code doesn't have any retry logic](https://github.com/frequenz-floss/frequenz-sdk-python/blob/2c8f61f6cab08d77a837fb0e730c4887ff678c4f/src/frequenz/sdk/microgrid/client/_client.py#L229-L244). By adding retry logic to the `ListComponents()` call, then I can't reproduce this issue any more, which seems to confirm the issue is the mock server taking too long to start.

Here is a patch to implement retry logic to `ListComponents`:

```diff
diff --git a/src/frequenz/sdk/microgrid/client/_client.py b/src/frequenz/sdk/microgrid/client/_client.py
index c075d2fd..4065557c 100644
--- a/src/frequenz/sdk/microgrid/client/_client.py
+++ b/src/frequenz/sdk/microgrid/client/_client.py
@@ -226,22 +226,47 @@ class MicrogridGrpcClient(MicrogridApiClient):
             AioRpcError: if connection to Microgrid API cannot be established or
                 when the api call exceeded timeout
         """
-        try:
-            # grpc.aio is missing types and mypy thinks this is not awaitable,
-            # but it is
-            component_list = await self.api.ListComponents(
-                microgrid_pb.ComponentFilter(),
-                timeout=DEFAULT_GRPC_CALL_TIMEOUT,  # type: ignore[arg-type]
-            )  # type: ignore[misc]
-        except grpc.aio.AioRpcError as err:
-            msg = f"Failed to list components. Microgrid API: {self.target}. Err: {err.details()}"
-            raise grpc.aio.AioRpcError(
-                code=err.code(),
-                initial_metadata=err.initial_metadata(),
-                trailing_metadata=err.trailing_metadata(),
-                details=msg,
-                debug_error_string=err.debug_error_string(),
-            )
+        retry_spec: RetryStrategy = self._retry_spec.copy()
+        error: grpc.aio.AioRpcError | None = None
+        component_list: microgrid_pb.ComponentList | None = None
+        while True:
+            _logger.debug("Making call to `ListComponents`")
+            try:
+                # grpc.aio is missing types and mypy thinks this is not awaitable,
+                # but it is
+                component_list = await self.api.ListComponents(
+                    microgrid_pb.ComponentFilter(),
+                    timeout=DEFAULT_GRPC_CALL_TIMEOUT,  # type: ignore[arg-type]
+                )  # type: ignore[misc]
+                break
+            except grpc.aio.AioRpcError as err:
+                error = err
+                api_details = f"Microgrid API: {self.target}."
+                _logger.exception(
+                    "`ListComponentData`: exception: %s api: %s",
+                    err.details(),
+                    api_details,
+                )
+
+            if interval := retry_spec.next_interval():
+                _logger.warning(
+                    "`ListComponentData`: connection ended, "
+                    "retrying %s in %0.3f seconds.",
+                    retry_spec.get_progress(),
+                    interval,
+                )
+                await asyncio.sleep(interval)
+            else:
+                _logger.warning(
+                    "`ListComponentData`: connection ended, "
+                    "retry limit exceeded %s.",
+                    retry_spec.get_progress(),
+                )
+                if error is not None:
[... 14 lines cut ...]
    tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_meter_data 
    ----------------------------------------------------------------------------------------------------------------------------------------------------- live log setup ------------------------------------------------------------------------------------------------------------------------------------------------------
    DEBUG    asyncio:selector_events.py:54 Using selector: EpollSelector
    ------------------------------------------------------------------------------------------------------------------------------------------------------ live log call ------------------------------------------------------------------------------------------------------------------------------------------------------
    DEBUG    grpc._cython.cygrpc:_server.py:60 Using AsyncIOEngine.POLLER as I/O engine
    DEBUG    frequenz.sdk.microgrid.client._client:_client.py:233 Making call to `ListComponents`
    DEBUG    frequenz.sdk.microgrid.client._client:_client.py:233 Making call to `ListComponents`
    DEBUG    frequenz.sdk.microgrid.client._client:_client.py:233 Making call to `ListComponents`
[... cut ...]
    WARNING  frequenz.sdk.microgrid.client._client:_client.py:383 `GetComponentData`, for component_id=83: connection ended, retrying (7/∞) in 0.011 seconds.
    DEBUG    frequenz.sdk.microgrid.client._client:_client.py:362 Making call to `GetComponentData`, for component_id=83
    PASSED                                                                                                                                                                                                                                                                                                              [ 40%]
    ```

shsms, 2023-09-20:
Ok,  maybe it is slowness.  But I wonder why there's slowness if there are two VMs.  Maybe they're forced to share the same CPU?  Also I have never seen this issue when running the tests locally.

**About retries:**

Just want to mention that the retries in the logs are expected, because the mock drops the connection after sending 3 values I think, but we try to receive a lot more.  This is just to test that the automatic reconnection to data streams work as expected.

The streaming methods have automatic retry, because the client exposes only a channel, for these methods, and there's no direct way for users of the client to be informed of such a failure.  Also the reconnection happens in a completely independent task.  

I'm doubtful about adding retry to non-streaming methods.  First it will block the caller until a connection is established, which might never happen.  But also if there's a failure,  the error gets propagated up immediately, unlike with the streaming methods, where the connections happen in a separate task.

We might still have to improve resiliency in higher layers,  like the `initialize` method, maybe.

llucax, 2023-09-20:
> I'm doubtful about adding retry to non-streaming methods. First it will block the caller until a connection is established, which might never happen. But also if there's a failure, the error gets [...]

Yeah, we definitely shouldn't do that just to make flaky tests pass, since I'm not familiar with the code I wasn't sure. What you say makes sense.

llucax, 2023-09-20:
> Ok, maybe it is slowness. But I wonder why there's slowness if there are two VMs. Maybe they're forced to share the same CPU? Also I have never seen this issue when running the tests locally.

Weird, I can reproduce locally more or less easily without having a high load. Having a base load of ~9% CPU usage, it takes me about 10 runs to have it failing. If I make the CPU work a lot [^1], I can reproduce it on every run (using `nox -R -s pytest_max -- --log-cli-level=DEBUG -x tests/microgrid/test_client.py`).

[^1]: I used this: https://www.baeldung.com/linux/cpu-spike-bash

    ```sh
    fulload() {
      dd if=/dev/zero of=/dev/null |
      dd if=/dev/zero of=/dev/null |
      dd if=/dev/zero of=/dev/null |
      dd if=/dev/zero of=/dev/null &
    }
    fulload; read; killall dd
    ```

llucax, 2023-09-26:
I noticed the first tests, which don't fail, use different port numbers for the grpc server, this is a workaround to make the tests non-flaky. I'm not sure why the pick-a-different-port approach wasn't taken for all tests and why it started failing only recently, as the tests seem to have been like this since this repo was created.

This is also a related commit 01728befc9d2c0891dc0fe75194a5a9ccd332638, where the graceful shutdown is added (or improved). It also seems like the timeout is super small.

We could increment that time-out, to give the server more time to shut down before we run the next test, but that's still inherently flaky, it just reduces the likeliness, while using an incrementing port will ensure the tests are not flaky. I will submit a PR with this change.

llucax, 2023-09-26:
On a second thought, the port thing should be solution, as the issue is not in the termination but in the waiting for the server to be really up. :thinking:

llucax, 2023-09-26:
BTW, now I'm not being able to reproduce the flakiness locally anymore. I wonder if something changed in the grpc dependency and this was an upstream problem.

llucax, 2023-09-26:
It seems like the flakiness mutated to an internal Python error :roll_eyes: 

```
tests/microgrid/test_client.py::TestMicrogridGrpcClient::test_charge Exception ignored in: <function BackgroundService.__del__ at 0x7f9e240be700>
Traceback (most recent call last):
  File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/src/frequenz/sdk/actor/_background_service.py", line 237, in __del__
    self.cancel("{self!r} was deleted")
  File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/src/frequenz/sdk/actor/_background_service.py", line 130, in cancel
    task.cancel(msg)
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/base_events.py", line 761, in call_soon
    self._check_closed()
  File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/asyncio/base_events.py", line 519, in _check_closed
Error:     raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed

INTERNALERROR> Traceback (most recent call last):
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/_pytest/main.py", line 271, in wrap_session
INTERNALERROR>     session.exitstatus = doit(config, session) or 0
INTERNALERROR>                          ^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/_pytest/main.py", line 325, in _main
INTERNALERROR>     config.hook.pytest_runtestloop(session=session)
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_hooks.py", line 493, in __call__
INTERNALERROR>     return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
INTERNALERROR>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_manager.py", line 115, in _hookexec
INTERNALERROR>     return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
INTERNALERROR>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_callers.py", line 152, in _multicall
INTERNALERROR>     return outcome.get_result()
INTERNALERROR>            ^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_result.py", line 114, in get_result
INTERNALERROR>     raise exc.with_traceback(exc.__traceback__)
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_callers.py", line 77, in _multicall
INTERNALERROR>     res = hook_impl.function(*args)
INTERNALERROR>           ^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/_pytest/main.py", line 350, in pytest_runtestloop
INTERNALERROR>     item.config.hook.pytest_runtest_protocol(item=item, nextitem=nextitem)
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_hooks.py", line 493, in __call__
INTERNALERROR>     return self._hookexec(self.name, self._hookimpls, kwargs, firstresult)
INTERNALERROR>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_manager.py", line 115, in _hookexec
INTERNALERROR>     return self._inner_hookexec(hook_name, methods, kwargs, firstresult)
INTERNALERROR>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_callers.py", line 152, in _multicall
INTERNALERROR>     return outcome.get_result()
INTERNALERROR>            ^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_result.py", line 114, in get_result
INTERNALERROR>     raise exc.with_traceback(exc.__traceback__)
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/pluggy/_callers.py", line 77, in _multicall
INTERNALERROR>     res = hook_impl.function(*args)
INTERNALERROR>           ^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/_pytest/runner.py", line 114, in pytest_runtest_protocol
INTERNALERROR>     runtestprotocol(item, nextitem=nextitem)
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/_pytest/runner.py", line 133, in runtestprotocol
INTERNALERROR>     reports.append(call_and_report(item, "call", log))
INTERNALERROR>                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/home/runner/work/frequenz-sdk-python/frequenz-sdk-python/.nox/ci_checks_max/lib/python3.11/site-packages/_pytest/runner.py", line 224, in call_and_report
[... cut ...]
INTERNALERROR>     astnode = ast.parse(content, "source", "exec")
INTERNALERROR>               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR>   File "/opt/hostedtoolcache/Python/3.11.5/x64/lib/python3.11/ast.py", line 50, in parse
INTERNALERROR>     return compile(source, filename, mode, flags,
INTERNALERROR>            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INTERNALERROR> SystemError: AST constructor recursion depth mismatch (before=117, after=167)
<actor_restart_limit> Resetting restart limit to None
```

llucax, 2023-10-05:
#699 seems to have improved things a lot. Maybe we can put this on hold now until we rewrite tests not to use the grpc mock server.

Marenz, 2024-03-07:
Can this be closed? The issue seems to be gone with the last fix mentioned.

llucax, 2024-03-07:
Yeah, let's do that, also this code was now moved to https://github.com/frequenz-floss/frequenz-client-microgrid-python/ and it shouldn't be too long before we completely get rid of the grpc mock in there.

closed 2024-03-07 by llucax