frequenz-floss/frequenz-sdk-python #671: Add references to other docs in tutorials

matthias-wende-frequenz, 2023-09-14:
1. To the minimum requirements (or to the whole installation section, as pointed out in a follow-up comment)
2. To some docs explaining how to use the sandbox (and eventually how to setup or buy a plan or something to get a "microgrid system" working)
3. An introduction to microgrid concepts (https://github.com/frequenz-floss/frequenz-sdk-python/issues/456)
4. Channels introduction (https://github.com/frequenz-floss/frequenz-channels-python/issues/177)

_Originally posted by @llucax in https://github.com/frequenz-floss/frequenz-sdk-python/pull/656#discussion_r1322819489_

matthias-wende-frequenz, 2023-10-17:
2) can be addressed when the sandbox api is finished.

matthias-wende-frequenz, 2023-10-17:
@llucax are you working on 4)?

llucax, 2023-10-30:
> @llucax are you working on 4)?

Yes