frequenz-floss/frequenz-sdk-python #704: Restructure timeseries module

matthias-wende-frequenz, 2023-09-27:
### What's needed?

Some of the parts should move out of the module because they are not solely related to timeseries.

### Proposed solution

## Quantities

* the quantity types

## Data pipeline

* logical meter
* grid frequency

## pools

* battery pool
* ...

### Alternatives and workarounds

Leave it as is.

llucax, 2023-09-28:
Yeah, imho everything that is working with the micrograms should be in the microgrid namespace.

Also I would move actors to their domain namespace, and leave `actor` for the actor framework stuff. This is also a source of a lot of cyclic imports.

llucax, 2024-07-29:
Closing in favor of #575.

closed 2024-07-29 by llucax