frequenz-floss/frequenz-sdk-python #724: PowerManager strategy for dealing with multiple actors with the same priority

shsms, 2023-10-13:
### What's needed?

The current approach when there are two actors with the same priorities is to order them based on their hashes,  which is the same as randomly assigning priorities to each of them one-time.

But there might be other possibilities:
- use the one with the narrower/conservative bounds,
- prefer discharge side bounds first rather than charge side bounds or vice versa,
- discard conflicting proposals with same priorities, instead of choosing one.

We probably need to get some real world experience to identify a way forward.

cc: @thomas-nicolai-frequenz

thomas-nicolai-frequenz, 2023-10-14:
More options could be:

* Time-Based Ordering: assigning priorities based on the time the proposals were received. The first one gets executed, creating a FIFO (First-In-First-Out) queue.
* We could combine multiple methods; for example, use narrower bounds when available but fall back to a time-based method as a tie-breaker.

thomas-nicolai-frequenz, 2025-03-25:
So we might have a few more choices but some aren't really applicable:

1. **Narrower/Conservative Bounds**: Prefer the actor whose Specified Limits are more restrictive (narrower bounds).
2. **Discharge vs. Charge Bias**: Optionally favor proposals on the discharge side over the charge side (or vice versa), depending on system requirements.
3. **Discard Conflicts**: Instead of selecting one, conflicting proposals at the same priority could be discarded.
4. **Time-Based Ordering**: Implement a FIFO approach where proposals are ordered based on their arrival time, so the first received gets priority.
5. **Hybrid Methods**: Combine strategies—for example, use the narrower bounds as the primary criterion, and if there’s still a tie, fall back to time-based ordering.

To me it feels we might one to support two of them from the beginning: My choice would be:

1. Random 
2. Narrower/Conservative Bounds
3. FIFO

I sense that in our use case where safety is top of mind critical—tie-breaking strategies that prioritize conservative limits (for example, selecting the narrower bounds) seem better suited then going for FIFO straight away. However, FIFO strength is simplicity, transparency and easy of use I guess.

thomas-nicolai-frequenz, 2025-03-25:
I would probably follow a suggestion laid out by good old AI. It seems straight forward and a strategy by prioritizing conservative bounds:

1. Primary Criterion – Conservative (Narrower) Bounds:
2. Secondary Criterion – Time-Based Ordering (Tie Breaker)

Here is a brief example explaining the idea:

```
Actor A: Specified Limits of -40 kW to 30 kW → Range = 70 kW.
Actor B: Specified Limits of -50 kW to 40 kW → Range = 90 kW.
```

In this case, Actor A is considered more conservative.

There is also a more illustrative example including not just bounds but also the power desired (requested):

Illustrative Scenario (Priority 2):

```
Actor A (2A):
Specified Limits: -40 kW to 30 kW (range = 70 kW)
Desired Power: 20 kW
Received at time T1

Actor B (2B):
Specified Limits: -50 kW to 40 kW (range = 90 kW)
Desired Power: 25 kW
Received at time T2

Outcome: Both proposals are added, so the raw sum is 20 kW + 25 kW = 45 kW. However, because Actor A’s bounds are narrower, the effective operating envelope for Priority 2 is set by Actor A’s Specified Limits. This means that, after applying these limits (and any necessary adjustments or ramping), the Aggregate Power is determined with Actor A’s conservative constraints in mind. If the bounds had been nearly identical, then the proposal received earlier (the one with the earlier timestamp) would be favored.
```

thomas-nicolai-frequenz, 2025-03-25:
**Conservative Bounds with a Stability/Confidence Metric**

One of the disadvantages of the proposal above is that its responsiveness is limited and the outcome is that the effective operating limits come from the actor with the better (narrower) fixed order. Another proposal would be to consider combining conservative bounds with a stability/confidence metric tie-breaker. This would lead favoring the actor with both narrower and stable (consistent) power proposals. This is something we might consider in the next step:

Illustrative Scenario (Priority 2):

```
Actor A:
Specified Limits = [-40 kW, 30 kW] (Range = 70 kW)
Desired Power = 20 kW
Stability Score: High (consistent over 5 cycles)

Actor B:
Specified Limits = [-50 kW, 40 kW] (Range = 90 kW)
Desired Power = 25 kW
Stability Score: Low (variable proposals)
Result:

Both proposals contribute, giving a raw sum = 20 kW + 25 kW = 45 kW. The stability metric 
reinforces the conservative selection: Actor A’s stable and narrower bounds take precedence. 
Actor B’s proposal is adjusted to fit within the effective limits defined by Actor A. The final 
Aggregate Power, after any necessary clipping and ramping, reflects a cautious operating 
envelope influenced by the actor with higher confidence.

```

merlin-esser-frequenz, 2025-03-31:
@thomas-nicolai-frequenz 

I get the idea, why you want to give the actor with more narrow bounds the higher priority and what you wrote down makes sense for me.

However, I still have problems with understanding the "specified limits" after reading through your example.
In case there are only two actors, as in the illustrative scenario, I get that one actor has to provide the leading bounds. But once there are actors with higher priority then the two actors we are comparing here, I dont think that we have to choose the one with the more narrow bounds.

What speaks against using the actor with the wider bounds as the high priority actor?
I know, that out of safety reasons it makes sense, but I think that we should be aware of any safety issues upfront and make sure, that any safety related applications surely have a defined priority, that can not clash with other actors.

So I understood the "specific bounds" more as an economical thing than a safety thing, and the Safety issue is targeted by the "System Limits".

What I am saying is, that if we always decide for the more narrow bounds, then we might limit ourself for no reason.

thomas-nicolai-frequenz, 2025-03-31:
> What speaks against using the actor with the wider bounds as the high priority actor?

SAFETY!