frequenz-floss/frequenz-sdk-python #727: Proposal life-time management in PowerManager's Matryoshka algorithm

shsms, 2023-10-16:
### What's needed?

We need a cleanup mechanism for old proposals under these situations:
  - when a new proposal for an existing priority+source_id is received with `None` values for both preferred power and bounds,
  - when the lifetime of a `BatteryPool` instance ends,

matthias-wende-frequenz, 2023-10-17:
> when the lifetime of a BatteryPool instance ends,

when the channel gets closed

shsms, 2024-01-18:
This is a bit urgent, because if one of the actors setting bounds goes away, its last preference would remain for ever.

I'm working on adding a timeout for every request sent to the PowerManager,  and that would transparently take care of this, because users are already having to repeat their commands, because after an gap they get dropped by the microgrid API otherwise.