frequenz-floss/frequenz-sdk-python #774: Update protobuf import to avoid pylint errors

daniel-zullo-frequenz, 2023-11-07:
### What's needed?

Remove the pylint `no-name-in-module` disable comments.

### Proposed solution

Update import to avoid the pylint error `no-name-in-module` as follows:

From:
```python
from google.protobuf.empty_pb2 import Empty  # pylint: disable=no-name-in-module
```

To:
```python
from google.protobuf import .empty_pb2
```

And then use `empty_pb2.Empty` instead

Once the change above is made remove all the comments to disable the pylint error in [`src/frequenz/sdk/microgrid/client/_client.py`](https://github.com/frequenz-floss/frequenz-sdk-python/blob/v1.x.x/src/frequenz/sdk/microgrid/client/_client.py)

### Additional context

Suggested in https://github.com/frequenz-floss/frequenz-sdk-python/pull/708#pullrequestreview-1717063060

llucax, 2023-11-07:
I would extend this issue to change all other imports from protobuf stuff, I think there should be more in the client code.

llucax, 2023-11-10:
OK, so it seems the test I did was wrong, and we effectively need to add an ignore where the symbol is used anyways, so I'll close this as invalid.

BTW, the underlying issue seems to be that `pylint` doesn't read stub (`.pyi`) files, which is supposed to be fixed in 3.0.0, but doing a simple test it seems like the issue persists with 3.0.0.

I guess we need to test it properly and maybe report to `pylint` if the issue is still there.

* https://github.com/pylint-dev/pylint/issues/6281
* https://github.com/pylint-dev/pylint/issues/4987

closed 2023-11-10 by llucax

llucax, 2023-11-24:
OK, it seems it is not really fixed in `pylint` 3, according to this message: https://github.com/pylint-dev/pylint/issues/6281#issuecomment-1571705142

> Unfortunately #4987 won't resolve this. #4987 makes it possible to lint .pyi files but it doesn't make it possible to check if a member is defined in a example.pyi file after the member is not found in the example.py file.