frequenz-floss/frequenz-sdk-python #784: Add and maintain whitelists for message streams from microgrid API

matthias-wende-frequenz, 2023-11-15:
### What's needed?

The microgrid API has changed and now the data streaming request sends a list of metrics.
With that change from a product type to a list type it becomes feasible to add filter and the proposed change in the microgrid-api [`ReceiveComponentDataStream`](ReceiveComponentDataStream) has been [suggested](https://github.com/frequenz-floss/frequenz-api-microgrid/issues/178).

In the current implementation, using the last stable microgrid api release, a request to a component returns a stream with a full set of metrics that belong to that component category.

In the future the call will need a whitelist of parameters that are should be returned.

### Proposed solution

There are several possibilities. 

1) We mimic the old behavior by adding a full set of metrics for a category but leave out unused metrics.

2) We only request what is needed and issue a new request to the api when a new datastream is established.

Option 2) is closer to the sdk design where we start a data pipeline only on demand to the `DataSourcingActor`
Furthermore it makes maintenance of the whitelist easier because we only get what has been asked for. If a user asks for a new metric it will seamlessly added to the stream.

llucax, 2023-11-16:
The link to [ReceiveComponentDataStream](https://github.com/frequenz-floss/frequenz-sdk-python/issues/ReceiveComponentDataStream) in the issue description is broken, not sure where you wanted it to point to.

llucax, 2024-04-09:
This will be done as part of the microgrid update:

* #844

closed 2024-04-09 by llucax