frequenz-floss/frequenz-sdk-python #785: Rename DataSourcingActor

matthias-wende-frequenz, 2023-11-15:
### What's needed?

When requesting a metric from a microgrid component one has to send a `ComponentMetricRequest` to the `DataSourcingActor`

We should rename this Actor to something that makes it immediately clear what it is doing especially w.r.t. to the upcoming API clients additions.

### Proposed solution

* MicrogridDatastreamActor
* MicrogridComponentDataActor
* ComponentMetricActor

matthias-wende-frequenz, 2023-11-15:
@llucax @shsms suggestions?

llucax, 2023-11-16:
The idea of this actor was that it gets metrics from different sources, that was the reason for the generic name. But we can review that idea now that new APIs are coming up if the abstraction doesn't really work.

llucax, 2023-11-16:
It is true that it is tied to component metrics, so it could be `ComponentMetricSourcingActor`. But the resampling actor is also tied to it. This also means if we have other data sources, we need also another resampling actor, which all brings me to the "I told you so" when I said we shouldn't tie the reampling actor to component metrics and just do the connection between the source and the resampling from outside of the actor instead of letting the actor do it itself :trollface:

shsms, 2023-11-16:
> It is true that it is tied to component metrics, so it could be ComponentMetricSourcingActor. But the resampling actor is also tied to it.

And continue calling it the data sourcing actor,  like we do with the [`ComponentMetricsResamplingActor`](https://github.com/frequenz-floss/frequenz-sdk-python/blob/ad298cdbb024c1d8708a09b9ec4e01bcc7677d0b/src/frequenz/sdk/actor/_resampling.py#L24)?  That sounds good to me.