frequenz-floss/frequenz-sdk-python #797: Update actor the documentation

daniel-zullo-frequenz, 2023-11-28:
### What happened?

Initially, the actor was designed to run continuously unless there was an error or it was intentionally stopped. Later, the actor was updated to support a new use-case where the actor exits the running loop upon a normal termination.

The current actor documentation and examples primarily focus on stopping the actor manually, with only a brief mention that the actor also naturally terminates upon completing execution without errors. Some examples are either outdated or not entirely accurate.

For example, here it documents how it is expected to be used, as continuously running actors:  https://github.com/frequenz-floss/frequenz-sdk-python/blob/v1.x.x/src/frequenz/sdk/actor/__init__.py#L214-L218
However, in this part it states that actors will be restarted only when it exits with errors: https://github.com/frequenz-floss/frequenz-sdk-python/blob/v1.x.x/src/frequenz/sdk/actor/__init__.py#L38-L40

Also for this example, the actor will block either until execution termination or until it gets stopped
https://github.com/frequenz-floss/frequenz-sdk-python/blob/v1.x.x/src/frequenz/sdk/actor/__init__.py#L70-L88

Also internal documentation needs to be updated to reflect what the code actually does.
For instance, the actor will also exit if it finishes execution successfully without any error but this is not reflected in the function
https://github.com/frequenz-floss/frequenz-sdk-python/blob/v1.x.x/src/frequenz/sdk/actor/_actor.py#L75-L111

### What did you expect instead?

Update or extend documentation and examples in actor class and module to reflect what the actor code does, including the scenario where the actor terminates execution normally.

### Affected part(s)

Actors or actors utilities (decorator, etc.) (part:actor), Documentation (part:docs)

llucax, 2023-11-30:
In https://github.com/frequenz-floss/frequenz-sdk-python/blob/471adc6623507eeb2f9fba693a62b735365b6261/src/frequenz/sdk/actor/__init__.py#L214 it says:

> Normally an actor should run forever (or until it is
> [stopped][frequenz.sdk.actor.Actor.stop]), so it is very common to have an infinite loop
> in the `_run()` method, typically receiving messages from one or more channels
> ([receivers][frequenz.channels.Receiver]), processing them and sending the results to
> other channels ([senders][frequenz.channels.Sender]).

Which doesn't look incorrect IMHO, it explicitly says one  can stop the actor manually. Do you mean we should just not mentioning that actors "normally should run forever", or that they could also stop on their own?

llucax, 2023-11-30:
https://github.com/frequenz-floss/frequenz-sdk-python/blob/471adc6623507eeb2f9fba693a62b735365b6261/src/frequenz/sdk/actor/__init__.py#L38-L40

This also doesn't say unhanded errors are the **only** way an actor can be restarted, it says it is **automatically** restarted on errors.

llucax, 2023-11-30:
https://github.com/frequenz-floss/frequenz-sdk-python/blob/471adc6623507eeb2f9fba693a62b735365b6261/src/frequenz/sdk/actor/__init__.py#L46-L48

Same here, this is only describing one feature, actors are async context managers and can be used like that, but you don't have to.

shsms, 2023-11-30:
>  Do you mean we should just not mentioning that actors "normally should run forever", or that they could also stop on their own?

I think we should definitely mention "run forever" as a pattern, but it is starting to become just _one_ usage pattern, because we are now exploring other patterns, like actors managing and stopping sub-actors, etc.

I think we need to make changes only once we know what those other patterns look like and there is atleast one implementation that we've refined and like, and want to share.

llucax, 2023-11-30:
So, in conclusion, after looking at the examples provided in the issue, I think the current docs are mostly correct, but I agree we could mention the case where an actor stops on its own (by just returning from the `run()` method normally) and add some example for that too.

llucax, 2023-11-30:
> I think we need to make changes only once we know what those other patterns look like and there is atleast one implementation that we've refined and like, and want to share.

This also looks like a reasonable path to me, I would spend a lot of time documenting patterns that we didn't use that much and maybe they don't work out that well after all. I mean, we could make it more explicit that there are other ways to use actors but maybe not spend a lot of time adding examples of other specific pattern just now.

llucax, 2023-11-30:
So I wonder if the priority high is really correct. Also @daniel-zullo-frequenz I think it would be helpful if you could submit a PR, even if it is a very rough draft, and maybe not even changing the text proper but just adding comments like "Here I would clarify X and Y" to have a better idea of how docs could look more clear to you.

daniel-zullo-frequenz, 2023-11-30:
> This doesn't look incorrect IMHO, it explicitly says one can stop the actor manually. Do you mean we should just not mentioning that actors "normally should run forever", or that they could also stop on their own?

They could also terminate on their own without calling stop(), that's missing

PR #883 (2024-04-02): Amend actor documentation
- [x] Update internal documentation for the actor method `_run_loop()`
- [x] Check and amend actor documentation and examples

Fixes #797
commit 1299db25 (2024-04-02): Amend actor documentation (#883)

- [x] Update internal documentation for the actor method `_run_loop()`
- [x] Check and amend actor documentation and examples

Fixes #797
closed 2024-04-02 by daniel-zullo-frequenz