frequenz-floss/frequenz-sdk-python #851: Split the `frequenz.sdk.actor` package to its own repository

llucax, 2024-01-29:
### What's needed?

We need to allow cloud applications to be able to write actors without depending on the microgrid.

### Proposed solution

Split the `frequenz.sdk.actor` package to its own repository:

- [ ] See which namespace would be possible to use for the new python package: `frequenz.actor` would be the natural choice, but it is still a namespace for the many actor packages. `frequenz.actor.core` or `frequenz.actor.base` might be some options, although not that clear. Keeping `frequenz.sdk.actor` could be another option, but we need to see if it doesn't conflict with `frequenz.sdk` living in another package too.
- [ ] Create the new repository and move the code there
- [ ] Release a v1.0.0 version (we want to ensure stability from day one, as we are stabilizing all the base repos).
- [ ] Remove the code from the SDK and depend on the new repo

### Use cases

All the cloud apps.

### Alternatives and workarounds

For now we've been copying protobuf files from the common API repo to the API repos that should depend on common to break the dependency, but this is clearly not sustainable.

### Additional context

This is a very complicated, multi-dimensional, problem, involving many parts. Is part (and one way) to solve the dependency conflict of the SDK depending on the microgrid API v0.15 (and common v0.4) and the new API clients depending on common v0.5.

shsms, 2024-01-30:
I like `frequenz.actor.base`, because it is the base class for all actors. but works badly for namespaced imports `from frequenz.actor import base`.

Maybe `frequenz.actor.actor_base`?

shsms, 2024-01-30:
it would be actor inception 3 levels deep:

```python
from frequenz.actor import actor

class A(actor.Actor):
```

llucax, 2024-02-01:
[image]

llucax, 2024-02-01:
[image]

I just did a test and we can use `frequenz.actor` :tada: 

As long as only one wheel is defining `frequenz/actor/__init__.py` it seems to be fine.

llucax, 2024-06-17:
This will actually be moved to the new [core](https://github.com/frequenz-floss/frequenz-core-python/) repository, as having a repo just for one class doesn't seem to have a lot of sense... Updating the issue title and description.

llucax, 2025-02-13:
> This will actually be moved to the new [core](https://github.com/frequenz-floss/frequenz-core-python/) repository, as having a repo just for one class doesn't seem to have a lot of sense... Updating the issue title and description.

This will probably not be possible, as **actor = task + channels** (even when the base class doesn't use channels, I guess eventually we'll want to tie them more explicitly), and we want to use `core` in `channels`, it will introduce a circular dependency. So I see 2 solutions:

1. Separate actor repository as it was the original plan
2. Not provide an `Actor` class and just use `Service` instead until we really need to make an Actor anything else than a background task, and only create a new repo (and class) after that happens.