frequenz-floss/frequenz-sdk-python #872: EV something

llucax, 2024-02-05:
@shsms will create a description

shsms, 2024-02-28:
**Progress update (from PR #887):**

1. `EVChargerPool`'s interface now corresponds to that of `BatteryPool`.  Individual EV charger control is no longer possible, unless the pool is initialized for a single component.
1. `PowerManager` can handle EV chargers - it sends target power for a set of EV chargers to the power distributor.
2. `PowerDistributor` can handle allocating power to EV chargers as soon as EVs get connected, if there's unallocated power left from the target power from the power manager.

**Pending items:**

1. `PowerDistributor` needs support for throttling EV chargers when they are consuming more power than latest target power coming from the power manager.
2. Testing/Unit tests

**Notes on throttling:**

But because we are dealing only in `Power` values and not `Current`, we can't do the optimizations for phase-rotation we did in the original actor, when just one phase has a breach.  This will still have the stage1(min charging power ~=6A) and stage2(stop completely) throttling, but will be much simpler otherwise.

And the target power has to be calculated outside.  But I'm sure a grid protection actor (like the one we already(~) have for batteries) that sets higher priority bounds through the PowerManager would be enough here too, when that's the requirement.

The PowerDistributor's distribution Algorithm for EV chargers is a modular self-contained thing now.  It is a simple first-come first-serve algorithm now, but can be swapped for other strategies as well.

For your attention, @thomas-nicolai-frequenz

shsms, 2024-02-29:
Now that we can have multiple PowerManager instances - one for each component category, some of the highest priority actors like a grid safety actor might need to set bounds on all such PowerManager instances, because each of them would affect the grid power, for example.

PR #900 (2024-04-03): Control methods for `EVChargerPool` using `PowerManager` and `PowerDistributor`
This PR adds new `propose_power` and `power_status` methods to the `EVChargerPool` similar to the `BatteryPool`.  These method interface with the `PowerManager` and `PowerDistributor`, which currently uses a first-come-first-serve algorithm to distribute power to EVs.

The following changes were made:
- An `EVChargerStatusTracker` implementation which tracks the health of an EV charger and whether an EV is connected.
- `PowerDistributor` and `PowerManager` are now modular to support multiple component types.
- Add  [... cut]
commit df875eed (2024-04-03): Control methods for `EVChargerPool` using `PowerManager` and `PowerDistributor` (#900)

This PR adds new `propose_power` and `power_status` methods to the
`EVChargerPool` similar to the `BatteryPool`. These method interface
with the `PowerManager` and `PowerDistributor`, which currently uses a
first-come-first-serve algorithm to distribute power to EVs.

The following changes were made:
- An `EVChargerStatusTracker` implementation which tracks the health of
an EV charger and whether an EV is connected.
- `PowerDistributor` and `PowerManager` are now modular to support
multiple component types. [... cut]
closed 2024-04-03 by shsms