frequenz-floss/frequenz-sdk-python #882: Add (re)active and apperent power to FormulaEngine

matthias-wende-frequenz, 2024-02-22:
### What's needed?

At the moment when calling a high level power function we always return the active component of the power. We also want to be able to use the logical meter high level interface to read the other power components power.

### Proposed solution

My suggested solution would be to add an optional parameter to the interface.

```python
grid = microgrid.grid(power_component = ComponentMetricId.REACTIVE_POWER)
```

and add this as a parameter to the [formula generator](https://github.com/frequenz-floss/frequenz-sdk-python/blob/v1.x.x/src/frequenz/sdk/timeseries/formula_engine/_formula_generators/_grid_power_formula.py#L15).

shsms, 2024-02-22:
Do we only need reactive power or do we need apparent power also?  And do we need it just for the grid?

I think it has to be a separate function `microgrid.grid().reactive_power`, because that has a different unit(VAR) and needs a different Quantity type.  Same for `apparent_power`, which is (VA).

shsms, 2024-02-22:
Also does it have to be per-phase or single 3-phase values?  did you have a usecase in mind for adding these, because that might tell us what the data should look like.

merlin-esser-frequenz, 2024-03-19:
@shsms This ticket has been created based on a usecase which doesnt really need the SDK and can be solved using the Reporting API or the Microgrid API.
However, for the future it could be useful to also read out all the meter and inverter values via the FormulaEngine. But currently there is no priority for it.

matthias-wende-frequenz, 2024-03-26:
First we need to support these metrics here -> https://github.com/frequenz-floss/frequenz-client-microgrid-python/blob/v0.x.x/src/frequenz/
client/microgrid/_component.py