frequenz-floss/frequenz-sdk-python #967: [MovingWindow] Add a trigger that fires after having received a fixed number of samples

matthias-wende-frequenz, 2024-06-10:
### What's needed?

Currently there is no clean way to wait until the MovingWindow got updated with a certain number of samples.
This feature has been requested in the past and there has been an initial implemented attempt https://github.com/frequenz-floss/frequenz-sdk-python/pull/387.

This feature has been requested again now by @cwasicki.

### Proposed solution

Introduce an asynchronous `wait_for_samples` method that returns once the requested number of new samples has been stored in the `MovingWindow`.

### Use cases

* replace a while loop in the `Forecast actor` (https://github.com/frequenz-floss/frequenz-sdk-python/pull/387#issuecomment-2152273322)

cwasicki, 2025-01-27:
It would be great if this could be part of v1.0 since we are still working around this with sleep calls. What was the issue with the implementation in the PR that was mentioned above?

FYI @phillip-wenig-frequenz

matthias-wende-frequenz, 2025-01-30:
Actually Channels could be seen as a convenience wrapper on top of condition objects.

matthias-wende-frequenz, 2025-01-30:
> It would be great if this could be part of v1.0 since we are still working around this with sleep calls. What was the issue with the implementation in the PR that was mentioned above?

AFAIR there is no issue. Someone has to pick it up.

shsms, 2025-01-30:
I can pick this up next week if no one is already looking into this.

cwasicki, 2025-01-30:
That would be great, I am pretty sure that no one is on it yet.

shsms, 2025-02-04:
#1159

PR #1159 (2025-02-12): Add a `wait_for_samples` method to the `MovingWindow`
Closes #967
commit bceccfc4 (2025-02-12): Add a `wait_for_samples` method to the `MovingWindow` (#1159)

Closes #967
closed 2025-02-12 by shsms