frequenz-floss/frequenz-sdk-python #985: [Documentation] `ComponentMetricId` is not documented

matthias-wende-frequenz, 2024-06-26:
### What happened?

The `ComponentMetricId` is used as a type for parameters in several places in the Documentation but the Type itself is not documented, which makes it hard to determine what to pass.

### What did you expect instead?

Documentation.

### Affected part(s)

Documentation (part:docs)

llucax, 2024-06-26:
So the class is documented, it is just not publicly exposed, right? So this issue is only about re-exporting it in https://github.com/frequenz-floss/frequenz-client-microgrid-python/blob/v0.x.x/src/frequenz/client/microgrid/__init__.py, right? Or do you think the class documentation needs to be improved too?

matthias-wende-frequenz, 2024-06-27:
It's about exposing it publicly.

llucax, 2024-06-28:
Easy fix then.

commit 99b25ec8 (2024-07-24): Expose `ComponentMetricId` in the public API (#990)

This was moved to the microgrid API client, so it wasn't exposed any
more in the SDK. This commit exposes it again.

Fixes #985.
PR #990 (2024-07-24): Expose `ComponentMetricId` in the public API
This was moved to the microgrid API client, so it wasn't exposed any more in the SDK. This commit exposes it again.

Fixes #985.
closed 2024-07-24 by llucax