frequenz-floss/frequenz-sdk-python #1024: Accounting for unknown (because of missing data) but probably existing power

shsms, 2024-08-06:
### What's needed?

When there's no data for some components, we should stop assuming that everything is `None`, because that could be very costly in big installations.

When PV inverters are not responding, we need to assume they're probably operating at maximum power.  

We need to have an estimate of what could be going on with these PV inverters when they're not sending data.  This should be available from the `PVPool`.

Then this needs to be extended to other pools.

llucax, 2024-08-06:
> When PV inverters are not responding, we need to assume they're probably operating at maximum power.

I would say what the assumption is should be left to be decided by the user, because different applications might need to make different assumptions. For example when shaving peaks, you probably want to assume no power is generated by missing PVs and discharge batteries more preemptively to avoid peaks as much as possible.

My suggestion is to make what the assumption is for missing data configurable per pool, so the user creating the pool can say this pool should assume a component is providing maximum power or not power depending on what its safety requirements are.

llucax, 2024-08-06:
This is also related to:

* https://github.com/frequenz-floss/frequenz-sdk-python/issues/1021

They are about different types of missing data (or data quality), I'm not sure if once this issue is resolved, #121 is still necessary. @cwasicki ?

cwasicki, 2024-08-06:
(fixed your link)

For the case of unreachable inverters this one might be sufficient, but I see #1021 useful to more problems with the data.