frequenz-floss/frequenz-sdk-python #1078: SoC does not get updated

daniel-zullo-frequenz, 2024-09-23:
### What happened?

An application kept reporting 100% SoC while the real SoC queried from the components was around 20%. It seems the application kept the outdated SoC as there wasn't any update for this metric. Apparently the SoC was no longer reported when new PV inverters were added to the microgrid (the PV inverters were reporting only Error state). Unfortunately there is no logs available.

### What did you expect instead?

The SoC is reported and it is up-to-date

### Affected version(s)

Observed in v1.0.0-rc601

### Affected part(s)

Microgrid (API, component graph, etc.) (part:microgrid)

### Extra information

The SoC was reported correctly once the application was restarted

llucax, 2024-09-25:
We observed this issue when the component graph was misconfigured as follows:

```
      grid_1
     /        \
meter_2    meter_3
```
`meter_3` was added by mistake and wasn't really a meter at all, so it was reporting no data. Because of this the SDK gave up on calculating the grid load.

This shouldn't affect how the SoC is calculated, but it might be relevant and some unexpected side-effect.

shsms, 2024-09-25:
We have tasks that calculate the SoC and stream it.  My theory is that the tasks must have crashed because of an exception.

The only real case I can see now is if both `soc_upper_bound` and `soc_lower_bound` are equal.  I'll add a fix for that.

https://github.com/frequenz-floss/frequenz-sdk-python/blob/2e7a09691db32ab48e68d77bfe5ac0d0d12d43c1/src/frequenz/sdk/timeseries/battery_pool/_metric_calculator.py#L400-L402

But I am continuing to look for other ways to reproduce.

closed 2024-10-10 by shsms

llucax, 2024-10-11:
We are just speculating about #1081 being the fix for this, right? I guess we can still leave it closed and reopen if it happens again.

commit 443e83e5 (2024-10-10): Improve resiliency of long-running async tasks (#1081)

Closes #1078
PR #1081 (2024-10-10): Improve resiliency of long-running async tasks
Closes #1078