frequenz-floss/frequenz-sdk-python #1125: PowerManager algorithm that can shift or limit bounds

shsms, 2024-12-04:
### What's needed?

Currently the power manager has a 2-group approach, where each group has a Matryoshka implementation that supports only limiting of bounds.  And the first group shifts the second group's bounds.

We need a more flexible implementation where each actor is able to decide whether they want to shift or limit the bounds of the lower priority actors.

The following changes are necessary to achieve this:

- The priority becomes optional.
- A proposal with just a target power and _without_ bounds would shift the other actors.
- A proposal with target bounds and optional target power would limit other actors with no priority or lower priorities.

llucax, 2024-12-04:
It might be good to add some examples on how we envision this new algorithm to work so we don't forget, as I'm not sure when we'll have time to actually implement it.

shsms, 2025-03-31:
We finally settled on an approach that always adds the powers set by actors, but also allows actors to limit the power set by lower priority actors.  This is implemented in the PR here: https://github.com/frequenz-floss/frequenz-sdk-python/pull/1146

closed 2025-04-08 by shsms